bastibl/gr-ieee802-11 #393: no data packet, just ack packets

mkk0719, 2023-08-15:
Dear bastibl,

I use a wifi to perform as AP, a PC as a client. And I use iperf to communicate between AP and client. Then I use usrp and wifi_rx.grc to receive data. But in wireshark, it can just capture ack packets and some broadcast packets, there is no data packet in wireshark list. 

I use another pc to run wireshark, and it can actually capture data packet. 

I run wifi_rx.grc on ubuntu 20.04lts, and I use maint-3.10 gr-ieee802-11.

So could you give me some advice?

bastibl, 2023-08-15:
The AP might use a different MCS for the data, which requires a better SNR (-> try to improve signal quality). Or the GNU Radio receiver only receives frames up to 1500 byte, maybe the data frames are larger (-> change constants in the code). Or the devices use non-11a/g for the data communication (i.e., larger bandwidths, MIMO, etc) (-> this doesn't work with this module). I'd guess it's the latter. In that case, you'd have to force 11a/g mode.

mkk0719, 2023-08-15:
Thank you for your reply.

I use [openwifi](https://github.com/open-sdr/openwifi) as AP, and use a PC as client.
 I run iperf3 server on openwifi and iperf3 client on PC to continue tcp connection. 
in AP
`iperf3 -s`
in client
` ./iperf3.exe -c 192.168.10.122 -t 1200 -l 1200`
I can see data packet in wireshark of PC.
[image]

I have looked through the config of openwifi and make sure that I have set 11a mode in channel 48 and bandwidth is 20Mhz.

I change `#define MAX_PAYLOAD_SIZE 1800 `in util.h and recompile the project. I also change the MTU size to 1300. 
But when I run wifi_rx.grc with usrp B210. I still can only get ack packets and beacon packets in wireshark wifi.pcap.

[image]

closed 2023-09-16 by bastibl