bastibl/gr-ieee802-11 #394: static assertion failed

xiota, 2023-09-12:
Attempting to build this on Arch Linux using [aur/gr-ieee802-11-git](https://aur.archlinux.org/packages/gr-ieee802-11-git) results in a "static assertion failed" error.  The maintainer of the aur package was able to [reproduce](https://aur.archlinux.org/packages/gr-ieee802-11-git#comment-933516) the error.

> I can reproduce the problem and getting the same error message. As can be seen from the logs, the build is failing not because of missing dependencies, but due to a compilation error ("static assertion failed"). git-blame tells me that the line triggering the error was changed in commit abed50bf7fb029032bd60e78948b5709aa060cfc (Fri Aug 4 13:09:57 2023 +0200), which is the current HEAD of the maint-3.10 branch.
>
> Building without this commit (HEAD^ == 965e2d474fafb3211b1eb5f464b6ddcaa5d2d694) succeeds...

<details><summary>Here is a build log.</summary>

```
-- The CXX compiler identification is GNU 13.2.1
-- The C compiler identification is GNU 13.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/sbin/pkg-config (found version "1.8.1") 
-- Found Volk: Volk::volk  
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'gmp'
--   Found gmp, version 6.3.0
-- Found GMP: /usr/lib/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/libgmpxx.so  
-- Found Boost: /usr/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.83.0") found components: date_time program_options system regex thread unit_test_framework 
-- User set python executable /usr/bin/python3
CMake Warning (dev) at /usr/lib/cmake/gnuradio/GrPython.cmake:21 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/gnuradio/GnuradioConfig.cmake:37 (include)
  CMakeLists.txt:78 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found version "3.11.5") 
CMake Warning (dev) at /usr/lib/cmake/gnuradio/GrPython.cmake:27 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/gnuradio/GnuradioConfig.cmake:37 (include)
  CMakeLists.txt:78 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonLibs: /usr/lib/libpython3.11.so (found suitable exact version "3.11.5") 
-- Found Git: /usr/sbin/git  
-- Extracting version information from git describe...
fatal: No names found, cannot describe anything.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Looking for xmmintrin.h
-- Looking for xmmintrin.h - found
-- Using install prefix: /usr
-- Building for version: gabed50b / 1.1.0git
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - not found
CMake Warning (dev) at /usr/lib/cmake/pybind11/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
[... 58 lines cut ...]
                 from /usr/include/gnuradio/block.h:17,
                 from /home/main-builder/pkgwork/src/gr-ieee802-11/lib/../include/ieee802_11/decode_mac.h:20,
                 from /home/main-builder/pkgwork/src/gr-ieee802-11/lib/decode_mac.cc:17:
/usr/include/fmt/core.h: In instantiation of â€˜constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = const gr::ieee802_11::Encoding; typename std::enable_if<PACKED, int>::type <anonymous> = 0]â€™:
/usr/include/fmt/core.h:1810:51:   required from â€˜constexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T& ...) [with T = {const gr::ieee802_11::Encoding, const int, const int}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {gr::ieee802_11::Encoding, int, int}]â€™
/usr/include/fmt/core.h:1828:18:   required from â€˜constexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<T>::type>::type ...> fmt::v10::make_format_args(T& ...) [with Context = basic_format_context<appender, char>; T = {const gr::ieee802_11::Encoding, const int, const int}]â€™
/usr/include/spdlog/logger.h:374:75:   required from â€˜void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const gr::ieee802_11::Encoding&, const int&, const int&}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]â€™
/usr/include/spdlog/logger.h:90:13:   required from â€˜void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const gr::ieee802_11::Encoding&, const int&, const int&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const gr::ieee802_11::Encoding&, const int&, const int&>]â€™
/usr/include/spdlog/logger.h:96:12:   required from â€˜void spdlog::logger::log(spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const gr::ieee802_11::Encoding&, const int&, const int&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const gr::ieee802_11::Encoding&, const int&, const int&>]â€™
/usr/include/spdlog/logger.h:158:12:   required from â€˜void spdlog::logger::info(fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const gr::ieee802_11::Encoding&, const int&, const int&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const gr::ieee802_11::Encoding&, const int&, const int&>]â€™
/usr/include/gnuradio/logger.h:173:23:   required from â€˜void gr::logger::info(const spdlog::string_view_t&, const Args& ...) [with Args = {gr::ieee802_11::Encoding, int, int}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]â€™
/home/main-builder/pkgwork/src/gr-ieee802-11/lib/decode_mac.cc:149:9:   required from here
/usr/include/fmt/core.h:1577:63: error: â€˜fmt::v10::detail::type_is_unformattable_for<const gr::ieee802_11::Encoding, char> _â€™ has incomplete type
 1577 |     type_is_unformattable_for<T, typename Context::char_type> _;
      |                                                               ^
/usr/include/fmt/core.h:1581:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1581 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1581:7: note: â€˜formattableâ€™ evaluates to false
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/build.make:104: lib/CMakeFiles/gnuradio-ieee802_11.dir/decode_mac.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:252: lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```
</details>

bastibl, 2023-09-15:
Any chances you try to build against the git wip 3.11 branch of GNU Radio? The maint-3.10 branch only works w/ the GNU Radio maint-3.10 branch. They might introduce breaking changes at any time in the dev branch of the next release, so it's not possible to guarantee that this is working with the dev branch.

xiota, 2023-09-15:
Built against gnuradio 3.10.7.0.  [PKGBUILD](https://gitlab.archlinux.org/archlinux/packaging/packages/gnuradio/-/blob/main/PKGBUILD?ref_type=heads)

bastibl, 2023-09-16:
This seems to be an issue with libfmt v10. You already have a work-around patch as part of your build:
https://gitlab.archlinux.org/archlinux/packaging/packages/gnuradio/-/blob/main/fmt-10.patch?ref_type=heads
Here, it seems to be similar. The `enum Encoding` needs some info how it should be formatted. I don't know about libfmt, but there seem to be many similar problems, e.g.: https://github.com/facebook/watchman/pull/1141/files#diff-9fd987324c3b1586ccf74a3c66cacfc8764dd912a86ccf14c559558470a482dfR26-R29
I tried to adapt that for this module. Could you please give it a try?

xiota, 2023-09-16:
<details><summary>Still not building.</summary>

```
-- The CXX compiler identification is GNU 13.2.1
-- The C compiler identification is GNU 13.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Found Volk: Volk::volk
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Checking for module 'gmp'
--   Found gmp, version 6.3.0
-- Found GMP: /usr/lib/libgmpxx.so
-- Using GMP.
-- Found MPLIB: /usr/lib/libgmpxx.so
-- Found Boost: /usr/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.83.0") found components: date_time program_options system regex thread unit_test_framework
-- User set python executable /usr/bin/python3
CMake Warning (dev) at /usr/lib/cmake/gnuradio/GrPython.cmake:21 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/gnuradio/GnuradioConfig.cmake:37 (include)
  CMakeLists.txt:78 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found version "3.11.5")
CMake Warning (dev) at /usr/lib/cmake/gnuradio/GrPython.cmake:27 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/gnuradio/GnuradioConfig.cmake:37 (include)
  CMakeLists.txt:78 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonLibs: /usr/lib/libpython3.11.so (found suitable exact version "3.11.5")
-- Found Git: /usr/bin/git
-- Extracting version information from git describe...
fatal: No names found, cannot describe anything.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Looking for xmmintrin.h
-- Looking for xmmintrin.h - found
-- Using install prefix: /usr
-- Building for version: gc431137 / 1.1.0git
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - not found
CMake Warning (dev) at /usr/lib/cmake/pybind11/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/cmake/pybind11/pybind11Tools.cmake:50 (find_package)
  /usr/lib/cmake/pybind11/pybind11Common.cmake:188 (include)
  /usr/lib/cmake/pybind11/pybind11Config.cmake:250 (include)
  /usr/lib/cmake/gnuradio/GrPybind.cmake:3 (find_package)
  python/bindings/CMakeLists.txt:25 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3
-- Found PythonLibs: /usr/lib/libpython3.11.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/include (found version "2.11.1")
CMake Error at /usr/lib/cmake/gnuradio/GrPybind.cmake:238 (message):
  Python bindings for mapper.h are out of sync
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:46 (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!
```
</details>

bastibl, 2023-09-17:
what a pain :-) next try?

xiota, 2023-09-17:
27c8fdf builds successfully.  Thank you.

closed 2023-09-17 by xiota