bastibl/gr-ieee802-11 #395: Tx is router，Rx is USRP

sarielkk, 2023-09-27:
Hello！ First of all, thank you for your work, it has been a great help to us. 

Is it feasible to use a router as Tx and USRP X310 as Rx to receive the wifi signal sent by the router? 
If it is feasible, can I just use the wifi_rx.grc file in the example directly?

Looking forward to your answer.

bastibl, 2023-09-27:
Yes, that's possible. But make sure to use 20MHz at theX310 and that the router uses 11g or 11a.

sarielkk, 2023-09-27:
Thank you for your answer. By setting it up as you said, can you ensure that the WiFi signal received by usrp only comes from the router?

> Yes, that's possible. But make sure to use 20MHz at theX310 and that the router sends 11g or 11a signals.

Thank you for your answer. 
By setting as you said, can we ensure the WiFi signal received by usrp only comes from the router?

bastibl, 2023-09-27:
If you don't conduct the experiments in a faraday cage, there is no guarantee that the devices doesn't pick up other frames. They would have to be filtered out afterwards, if needed.

sarielkk, 2023-09-28:
> 如果您不在法拉第笼中进行实验，则无法保证设备不会拾取其他帧。如果需要的话，之后必须将它们过滤掉。

Thanks for your reply, I will try as you said.

sarielkk, 2023-09-28:
> If you don't conduct the experiments in a faraday cage, there is no guarantee that the devices doesn't pick up other frames. They would have to be filtered out afterwards, if needed.

Thanks for your reply, I will try as you said.

closed 2023-09-28 by sarielkk