bastibl/gr-ieee802-11 #397: Port is not connected

DuDafang, 2023-10-06:
I run the following command in the example folder: then I open wifi_phy_hier in gnuradio software, but unfortunately I still get an error, mainly
Value "ieee802_11.signal_field()" cannot be evaluated:
name 'ieee802_11' is not definedParam - Value(value):
Value "ieee802_11. LS" cannot be evaluated:
name 'ieee802_11' is not definedParam - Formatter Object(header_formatter):
Value "header_formatter.formatter()" cannot be evaluated:
'NoneType' object has no attribute 'formatter'Param - Value(value):
Value "ieee802_11. BPSK_1_2" cannot be evaluated:
name 'ieee802_11' is not definedParam - Algorithm(algo):
Value "chan_est" cannot be evaluated:
name 'chan_est' is not definedParam - Encoding(encoding):
Value "encoding" cannot be evaluated:

name 'encoding' is not defined

I also opened the wifi_loopback in the gnuradio software with the error: Port is not connected.

I'm sorry, I just touched these, is there something wrong with my operation? Thank you very much.

bastibl, 2023-10-06:
What command did you run? looks like you didn't paste it. Please see the readme for installation instructions. If you have further problems, please provide more information about your installation.

DuDafang, 2023-10-07:
Thank you very much for your reply. What does 'paste' mean?

With gnuradio installed, I executed the following commands in full:

git clone https://github.com/bastibl/gr-foo
cd gr-foo
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

git clone https://github.com/bastibl/gr-ieee802-11
cd gr-ieee802-11
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

I then went to the example folder and proceeded to execute the following command:
cmake .
make

After the above operations, gnuradion has found the Foo block, IEEE802.11 block.

bastibl, 2023-10-08:
I'm not sure if your problem is solved or if you still face the initial issue. But in case: What version of GNU Radio did you install? How did you install it? Were there any errors executing those commands? Did you set the environment variables as indicated in the readme? Did you make sure that the branch of this module matches the GNU Radio version? From the readme:
> Like GNU Radio, this module uses maint branches for development. These branches are supposed to be used with the corresponding GNU Radio branches. This means: the maint-3.7 branch is compatible with GNU Radio 3.7, maint-3.8 is compatible with GNU Radio 3.8, etc.

closed 2024-07-09 by bastibl