bastibl/gr-ieee802-11 #398: wifi_loopback.grc - flowgraph stops with waning messages

f-zumegen, 2023-10-11:
Hi Bastian,

I run into a problem when executing the loopback example to verify that I installed everything correctly. From my understanding, I follow all the steps described in the repo readme until "Checking your installation". When I load and execute wifi_loopback.grc inside gnuradio-companion, this is the console output:

```
Loading: "/home/fzumegen/gr-ieee802-11/examples/wifi_loopback.grc"
>>> Done

Generating: '/home/fzumegen/gr-ieee802-11/examples/wifi_loopback.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/fzumegen/gr-ieee802-11/examples/wifi_loopback.py

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Qt GUI: Could not restore geometry: restoreGeometry(self, Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'NoneType'
MAPPER: encoding: 0
chunks_to_symbols :info: set_min_output_buffer on block 5 to 207744
fft_v_fftw :info: set_min_output_buffer on block 3 to 2077440
ofdm_cyclic_prefixer :info: set_min_output_buffer on block 5 to 2077440
ofdm_carrier_allocator_cvc :info: set_min_output_buffer on block 6 to 207744
chunks_to_symbols :info: set_min_output_buffer on block 7 to 415488
tagged_stream_mux :info: set_min_output_buffer on block 8 to 207744
flowgraph :warning: Could not find port out in:
flowgraph :warning:   system

>>> Done (return code -11)
```
As a result, no GUI shows up and I don't think any Wifi packets are exchanged in loopback. Can you help me get this running after all?

Best
Frederik

bastibl, 2023-10-12:
Such things are usually a version issue. Please check the branch (=version) of this module with your GNU Radio version (see readme). If that's not the case, make sure that there are no artifacts from previous installations or installation attempts on your system.

f-zumegen, 2023-10-12:
Thanks for your reply.

I’m using the maint 3.10 branch and GNU radio is 3.10.x. I will try a new install.

closed 2024-07-09 by bastibl

vanwayplus, 2025-03-31:
Hi I have the same problem. My gnuradio is 3.10.7, and I downloaded the 3.10 branch. Should I try a lower version?