bastibl/gr-ieee802-11 #399: capturing only 2 out of 4 in handshake

yuval174, 2023-11-28:
hello and thank you for your project. I'm having a problem and I hope you could help me.
my setup is:
hackRF for SDR(i added DC filter block from the IQ correction library)
for host computer I tried both on new laptop with i7 and dragonOS. And intel nuc with i7 8th gen with ubuntu 22.04
the network I'm trying to capture is: android phone for hotspot and laptop connecting to him. also the other way around.

I'm using the wifi_rx.grc(with the rx_demo.sh so wireshark is active) and trying to capture the network traffic, and I'm getting almost only ACK and clear-to-send, with some QoS frame. and also the weird part(for me at least) is that i can only get **message 2 out of 4 and 4 out of 4 every time** I'm recording the handshake process. i do get many "O" error, so maybe this is related but i cant seem to get rid of them.
also I'm not sure if related or not but when I'm using the wifi_tx.grc I'm getting a lot of "U"

thank you for your time, sorry if it's not relevant for here.

yuval174, 2023-12-04:
hey thank you for the response.

i tested the hackRF in a vacant channel and there were indeed still overload so it seems that my DC filter isn't good enough, if you have any suggestion I will appreciate it, I tried 2 different blocks from the core lib. 
for now i switched to the BB60c from Signal Hound as a source, and in quite channel there are no overload error and no frames at all. switching to 5Ghz and forcing the hotspot to work in single channel seemed to work. now i get the full communication at least in regards to the hand shake. 
i think it is safe to say that in regard to your project the issue is resolved so thank you for the help.

I think I would want to try and expand the project to include also 802.11b. could you help me understand what it would require? I saw that the main difference is OFDM vs DSSS.I think i saw other gnu-radio lib that decode DSSS so maybe a mix and match is possible?
I don't have a strong background in programming, but i do have knowledge in signal processing and RF

bastibl, 2023-12-04:
I never looked into 11b. But there is this project: https://github.com/hui811116/gr-wifi-dsss (I never tried it). And, I remember, there was a 11b receiver for GNU Radio around 2010 maybe. But, I think, at that point CPUs were too slow to do the DSSS correlation. So they did it on an FPGA or DSP. But this should now be possible.

yuval174, 2023-12-04:
thank you for the help, if I will manage to create something that operate consistently I will update