bastibl/gr-ieee802-11 #400: Need help in generating wifi Becons using HackRF (wifi_tx.grc)

ShivaAlethea, 2024-01-22:
Hi, Im new to this GNU Radio.
Can you guys please help me to generate .grc for **Beacon** geneartion (using HackRF One).
I tried to ran [wifi_tx.grc](https://github.com/bastibl/gr-ieee802-11/blob/maint-3.10/examples/wifi_tx.grc), It is working.
But it is not generating any Becons.

Also tried to follow https://user-images.githubusercontent.com/77594215/106764041-59bd3400-662f-11eb-8a87-994884fa0361.png

But im not able to get the values that i need to pass on **header_formatter**, **Packet_Header**, **OFDM_carrier_Allocator** and **FFT**.

Can anyone help me to create this/new **.grc** for beacon generation.

Thank you

bastibl, 2024-01-22:
From the screenshot I only see that something is red, so something goes wrong. the error is not on the screenshot, so I can only guess:
- make sure you use the correct version/branch of this module as indicated in the readme
- check cmake output, if something is missing and, therefore, stuff is not built

See this [blog post](https://www.bastibl.net/gnuradio-wlan-scapy/) and this [talk and slides](https://archive.fosdem.org/2019/schedule/event/gr_scapy/) for how to inject frames.

ShivaAlethea, 2024-01-23:
Hi @bastibl Thank you very much for the reply.
With the help of above [blog post](https://www.bastibl.net/gnuradio-wlan-scapy/) created **beaconFrameGeneration.py** and **wifi_beacon_tx.grc**
[wifi_tx.zip](https://github.com/bastibl/gr-ieee802-11/files/14021990/wifi_tx.zip)

Running **wifi_beacon_tx.grc** is not giving any error, But still I'm not able to see the Beacons.
Im i missing anything here, please help.
[image]

Beacon frame generation using scapy:
[image]
[image]

GNU radio .grc (please find the attached zip for .grc file)
[image]

ShivaAlethea, 2024-01-24:
@bastibl 
Also tried the first scenario, No errors now.
Need a info >> on **FFT** >> **Window** (formula something like>> tuple(1/52**0.1.....))
[wifi_tx_1.zip](https://github.com/bastibl/gr-ieee802-11/files/14035244/wifi_tx_1.zip)
[image]
[image]

ShivaAlethea, 2024-01-25:
Hi @bastibl,
Tried to follow https://github.com/bastibl/gr-ieee802-11/issues/236 and observed below.
- If i try to capture PCAPs directly from the **Socket PDU**, It is working fine and I'm able to see some **Beacons**.
- But the same is **not working** with **WiFi PHY Hier**,  No packets are captured.

PCAP With >> Socket PDU - **Working**
[image]

PCAP With >> WiFi PHY Hier - **Not working**
[image]

**If available can anyone please share me the ".grc" for this.**

ShivaAlethea, 2024-02-07:
@bastibl Thanks for the support. [It](https://private-user-images.githubusercontent.com/157352391/298897148-7d937d73-f91c-4363-a693-286571d90734.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY3NzkzNjQsIm5iZiI6MTcwNjc3OTA2NCwicGF0aCI6Ii8xNTczNTIzOTEvMjk4ODk3MTQ4LTdkOTM3ZDczLWY5MWMtNDM2My1hNjkzLTI4NjU3MWQ5MDczNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwMVQwOTE3NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xM2U0ZWI4NzI0NDlkNzVjNGQ0OWVlOWIwNDRmYmI2YmFhYjk2NmQ1ODZjNGNjZmRiNDQ4Y2U3NDA2OGY2YmEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.va3HMEA5GUoIBlYHcvHo7zF4FVPGhMRefRxuMhvHKxg) started working fine after HackRF Firmware update.
Now im able to observe my SSID's in laptop.

ShivaAlethea, 2024-02-07:
@bastibl 
Need one more clarification,
Is it possible to send multiple beacons from different ports and frequencies ?
Something Like:
[image]

**Trying to run this test is giving me :** RuntimeError: resolve_port: hierarchical block 'sink_impl': input 1 is not connected internally

```
File "/home/alethea/shiva/HackRF/gr-ieee802-11/examples/playback_test.py", line 200, in <module>
    main()
  File "/home/alethea/shiva/HackRF/gr-ieee802-11/examples/playback_test.py", line 180, in main
    tb.start()
  File "/usr/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 116, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
RuntimeError: resolve_port: hierarchical block 'sink_impl': input 1 is not connected internally

```
Anything else needs to be taken care here?

bastibl, 2024-02-07:
This not possible with the HackRF.

ShivaAlethea, 2024-02-08:
@bastibl Thank you very much for your support.
Closing this issue as it started working for me after firmware update.

closed 2024-02-08 by ShivaAlethea