bastibl/gr-ieee802-11 #402: Becacon and eapol receive error

georgeteht, 2024-02-22:
Hello, I am saving wifi packets using a usrp n210 device and using the sample grc file you shared. But I can't see beacon packets. What do you think could be my problem? When I switch to external wifi 5g channels, it does not save packets, why can this happen?

Can I also collect eapol packages using this library?

[image]

bastibl, 2024-02-22:
The receiver works only with 11a/g (i.e., OFDM 20MHz) WLAN frames. Maybe the AP uses 11b for the beacons.

georgeteht, 2024-02-22:
I don't just see beacon packets, but I also see EAPOL packets. Could it be that it doesn't come due to the package size? 
The size of incoming packages is about:35
The size of the beacon package is approx: 300

Also, can the system operate at a bandwidth of 40 mhz-80 mhz? Is it enough to just change the samp_rate value for this? (The usrp I am currently using does not support this, but how can it be done when using a suitable device?)

My accesspoint beacon packet details :

Tag: Supported Rates 1(B), 2(B), 5.5(B), 11(B), 9, 18, 36, 54, [Mbit/sec]
        Tag: DS Parameter set: Current Channel: 5
        Tag: Extended Supported Rates 6, 12, 24, 48, [Mbit/sec]
        Tag: AP Channel Report: Operating Class 32, Channel List : 1, 2, 3, 4, 5, 6, 7,
        Tag: AP Channel Report: Operating Class 33, Channel List : 5, 6, 7, 8, 9, 10, 11,
        Tag: Traffic Indication Map (TIM): DTIM 0 of 1 bitmap
        Tag: ERP Information
        Tag: HT Capabilities (802.11n D1.10)
        Tag: HT Information (802.11n D1.10)
            Tag Number: HT Information (802.11n D1.10) (61)
            Tag length: 22
            Primary Channel: 5
            HT Information Subset (1 of 3): 0x00
            HT Information Subset (2 of 3): 0x0004
            HT Information Subset (3 of 3): 0x0000
            Rx Supported Modulation and Coding Scheme Set: Basic MCS Set
        Tag: Extended Capabilities (1 octet)
        Tag: RSN Information
        Tag: Vendor Specific: Microsoft Corp.: WMM/WME: Parameter Element
        Tag: QBSS Load Element 802.11e CCA Version
        Tag: Vendor Specific: Ralink Technology, Corp.

bastibl, 2024-02-22:
If longer frames are not received that suggests that the SNR is bad. Check antenna, try different gains etc etc. IIRC, the N210 has an uncompensated filter at 20Mhz. It should print to console. So you could try 25MHz and down sample to 20MHz in GNU Radio. The receiver is for 11a,g,p only. Your card can send this mode, but the information you posted doesn't show what it does. How the beacon is encoded might also depend on the driver.

georgeteht, 2024-02-23:
How can I check them ?

bastibl, 2024-02-23:
For example by receiving in monitor mode w/ a normal device and check radio tap header.

closed 2024-07-09 by bastibl