bastibl/gr-ieee802-11 #403: gr-ieee802-11for bladerf?

liiso7, 2024-02-26:
Does this work for bladerf2.0? Why did I run wifi_tx.grc and wifi_rx.grc without producing any results（Replaced the USRP source/sink modules with osmocom source and sinks
respectively.）

liiso7, 2024-02-27:
1.wifi_loopback.grc works, and here's what I got:
[image]
2.. I use wifi_tx.grc to send data packets, and use monitor mode wireless network card to receive data at the same frequency. The receiver wireshark did not see any data packets sent by wifi_tx.grc, and wifi_tx.grc generated file wifi.bin without any content. Here is my wifi_tx.grc flowchart:
[image]
3. I use wifi_tx.grc to send packets and wifi_rx.grc to receive packets at same frequency, but it does not produce any results, here is my wifi_rx.grc flow diagram:
[image]

bastibl, 2024-02-27:
That `wifi.bin` is empty is as expected. At this output received frames would show up, but this is a TX flowgraph. The rest looks fine. I'd suggest to test different parameters (esp. gain), check all parameters again, and maybe add more padding in the PacketPad 2 block at the transmitter.

liiso7, 2024-03-11:
Now I can receive a small number of data frames sent by wifi_tx.grc through the wireless network card, but the sequence number of data frames is not consecutive, that is to say, packet loss occurred in the transmission process, may I ask what causes this? What is the data transmission packet loss rate for this project?

bastibl, 2024-03-11:
Hard to tell. This can have many causes. Interference, gain too high, gain too low, wrong antenna, antenna at wrong port, pc too slow, SDR has an uncompensated DC offset, ... any many more. I've seen all of this in the past.

closed 2024-07-09 by bastibl