bastibl/gr-ieee802-11 #410: Failed to receive 802.11g wireless signal

xyexploring, 2024-07-01:
The MAC address of device A is 3c:6d:66:01:63:51, and it is connected to a router whose MAC address is 18:f2:2c:8b:99:f2. Now device A only supports 802.11g and sends a signal to another device in the same network segment, but it only receives the signal sent by the router to device A and cannot receive any signal sent by device A.    why？
[image]
[image]

bastibl, 2024-07-02:
You could try different gains and check that the device really sends 11g (and not 11b, for example). Some 11g frames also use a special preamble to be backwards-compatible to 11b. This preamble is also not detected. It is usually hard (or not possible) to change these parameters. Maybe the easiest would be to try a different card.

xyexploring, 2024-07-02:
Thank you very much for your reply

closed 2024-07-09 by bastibl