bastibl/gr-ieee802-11 #411: updation needed for IEEE 802.11n packet capturing

Bala26TN, 2024-07-08:
Greatings,
this project has helped me a lot, i would like to know how can this be updated for capturing all modes of WIFI that include IEEE 802.11(n/ac/ax) also, can u help me with the updated and extra blockes to be added for doing the same.
thanks in advance.

bastibl, 2024-07-08:
Very happy that the project helped you. Extending it for other standards would be a major effort, as they differ in bandwidth, preamble, codes (e.g., LDPC vs convolutional codes), MIMO, etc. etc. Also doing >=80MHz and higher MIMO orders on the CPU, might be hard. If you want to give it a try, I'd generate a reference frame (for example with the MATLAB WiFi Toolbox) and use the standard to come up with code to decode the reference frame.

Bala26TN, 2024-07-08:
thank you for the response
where can i get the code where u have generated the reference frame and decoded it.
basically i have to capture all WIFI packets that are present in the surrounding. but i am missing those devices having WIFI IEEE802.11n that why i have posted the query.

closed 2024-07-09 by bastibl

Cobi-Coder, 2025-04-29:
@Bala26TN Have you completed the 802.11n project with SDR?