bastibl/gr-ieee802-11 #414: WiFi Decode MAC removes FCS

Dowafu, 2024-09-11:
Hello,

I used Scapy to generate the Wi-Fi frames and send them through the loopback flow. When I generate *Dot11FCS* frames, the received messages no longer match the messages generated by Scapy, and in fact, the Frame Check Sequence gets removed. I am comparing against mac_out.

I tried to trace the behavior and found that decode_mac.cc line 155 truncates the FCS after checking it.

My expectation was that the loopback retains the FCS, and for captures with Wi-Fi cards, I have seen so far the FCS in the pcap. Do I have it the wrong way around and removing the FCS is the expected behavior?

## Example

```python
from scapy.all import *

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
frame = Dot11FCS(subtype=8, type=2, addr1='ff:ff:ff:ff:ff:ff', addr2='ce:6a:74:4d:85:48', addr3='ff:ff:ff:ff:ff:ff')/\
    Dot11QoS(TID=5,Ack_Policy=1,EOSP=0,A_MSDU_Present=0,TXOP=0)/Raw(b'BBBBBB')
frame_build = frame.build()
print(bytes.hex(frame_build))
sock.sendto(bytes(frame_build), ("127.0.0.1", 52001))
```

bastibl, 2024-09-12:
Hmm, I see that this can be unexpected. When implementing it, I didn't look at it from that perspective. I guess, it was just more natural to implement it that way. On TX side, the MAC sets the CRC; on RX, the decoder checks and removes it. You're right, it might have been better to keep the CRC, but changing it now would probably break some code :-/

Dowafu, 2024-09-19:
Thanks for the explanation. I have readded the FCS locally for my use case, as I use it only for TX and debugging my setup via the loopback and thus don’t mind breaking some code somewhere else 😉

closed 2024-09-19 by Dowafu