bastibl/gr-ieee802-11 #418: talk between two instances?

Brackii, 2024-11-04:
Hello,

Starting with a disclaimer. Don't really know if this is a good place to ask this question but I tried to use your library so you might know something good.

I want to use this create "wifi" traffic for a disturbance environment. From what I have read, it is really hard to do wifi with SDR. Can I connect two transceiver and talk "wifi" insde gnuradio and just send the TX out on two different SDRs?

Thanks in advance

bastibl, 2024-11-07:
I am not sure, if I fully understood the question, but you can send and receive WLAN frames with SDR. Either between two SDRs or in combination with commercial WLAN cards. It is, however, not possible to fully associate to an AP or have a standard compliant MAC layer (i.e., it doesn't send ACKs or do CSMA etc.).

Brackii, 2024-11-07:
Sorry for being a bit unclear, I'm new  to SDRs and my signals courses are a bit back in history... I need to get a grip of the vocabulary. 

I want to send frames within a flowgraph, but also send them out on HW sinks. I don't need to react on the frames sent from HW and I wont be receiving anything, it will be used to generate a disturbance into a test bed. I have created OFDM signals and transmit them, but I was struggling with creating a timing control that would simulate different throughputs. That's when I found your module (block? package?) and thought I might be able to use that. But it sounds like I need to implement something that would send ACKs for me.

bastibl, 2024-11-07:
If you create the frames on-the-fly, you want get precise timing and the frame generation would be too slow to saturate the channel.

I would, therefore, recommend to first write the samples to a file, i.e., replace the HW block at the end of the TX flowgraph with a file sink. The Packet Pad block just before the sink inserts a given number of samples before and after every frame. Configuring these numbers, you can control the inter-frame space and, therefore, the frame rate/throughput.

Then create a flowgraph with  just a file source that is connected to the HW sink.

ACKs do not seem to be relevant for your application.

Brackii, 2024-11-08:
Aah, okay, so even if you keep the triggers (I was thinking tags)  within the software it will be to slow? Would it be to slow if you had the IQ data stored in a vector/stream and just muxxed them out depending on sample count? Because I have actually already stored the frames, but I was storing a continuous transmission with out gaps and then I was failing on chopping them up when I wanted to send it (bunch of underruns, late packets and so). Thanks for the idea of using the packet pad, seems so obvious now that you say it. Will try that to start with. Big thanks!

bastibl, 2024-11-08:
I don't know if it will be too slow, since I don't know your HW and the number of frames per seconds that you want to transmit.
You can also store a IQ samples in a vector source, but with files the process is easier. Again, just use the transmit flowgraph to generate an IQ file instead of streaming the frames to HW. Then you also get padding between frames.

Brackii, 2024-11-08:
Of course. 

Will do. Thanks your time!

closed 2024-11-08 by Brackii