bastibl/gr-ieee802-11 #421: Decoding failures with HackRF One using wifi_rx.grc

Sun-Sida, 2025-02-24:
Hello,

I'm running gr‑ieee802‑11 on HackRF One at a fixed 20 MHz sample rate, trying to decode some wifi2.4 chan1 or7. Sadly, yet I never get successful MAC frame decoding. The decode MAC block’s debug messages report frames being dropped because they're “too large (symbols or bits)” or failing checksum validation. I also see multiple warnings like “starting to receive new frame before old frame was complete,” which suggests overlapping frames or an overwhelmed processing chain.

Furthermore, my Constellation Diagrams are inconsistent—sometimes I see a central dot, symmetric two-point, or even a 16‑point pattern, but no stable constellation diagram emerges. I do not really know what a good Constellation Diagram should look like.

Any insight on what might be causing these symptoms and how to adjust parameters to achieve reliable decoding would be appreciated!

Thank you!

[image]

[image]

[image]

[image]

bastibl, 2025-02-27:
I'm not sure, I understand the problem. When you receive from multiple devices, they'll use different encodings. The BPSK and 16-QAM that you posted look good. Also an AP might use BPSK for transmitting beacons and 16-QAM for unicast to a client. 

Sometimes the receiver might wrongfully sync on noise, which will result in a dot in the constellation diagram. This can happen more often, if the threshold in sync short is too low or the DC offset compensation is not great. Every frame/PDU that is output by "WiFi Decode Mac" is correctly received. I'd expect the frames from the screenshots of the constellation diagrams to be decodeable.

Sun-Sida, 2025-03-01:
Hello! Thanks for your reply. And sorry I did not make it clear. I mean I did not really know if my Constellation Diagrams looks good enough since I could never get successful MAC frame decoding. 

Maybe some parameters should be adjusted, but I don't detailly know how to adjust them. I tried to changed Gains but did not really help. And I could see some "received complete frame - decoding" but sadly always ending up with "checksum wrong -- dropping". And there are quite a few warning “starting to receive new frame before old frame was complete”. And I am okay with the "O" overflow problem, just a few sometimes.

Do you have any idea about this? Thank you!

[image]

bastibl, 2025-03-03:
Try connecting the output of `WiFi Parse MAC` to the print input of the `Message Debug` block. Also, use the loopback flowgraph to check if your instrumentation works and what is printed in case of success. Maybe you expect something that doesn't happen. Also if you enabled the file sink, you could easily check, if packets will be output.

Sun-Sida, 2025-03-07:
Thanks so much for your advice! After modifying some Gains, I successfully decoded some frames. Thank you for this wonderful repo!

closed 2025-03-07 by Sun-Sida

Cobi-Coder, 2025-03-24:
[image]

[image]

I adjusted the gain but the constellation diagram is not as expected, I don't have the Correct IQ block

Cobi-Coder, 2025-03-31:
> Thanks so much for your advice! After modifying some Gains, I successfully decoded some frames. Thank you for this wonderful repo!

Can you help me

bastibl, 2025-03-31:
The block is available in this repository: https://github.com/ghostop14/gr-correctiq.